mahjongg2/ZX81plus38 #1: ZX80 compatibility

stevexyz, 2020-06-29:
Does it work also as a ZX80?
Here you can see a nice trick that will do with 4 resistors in case!
https://sinclairzxworld.com/viewtopic.php?t=1329

mahjongg2, 2020-06-30:
Never tried to run Z80 software on a ZX81, but I see no reason why it couldn’t work, if you use a ZX80 ROM, and add the pull down resistors, just try it!

let me know how it went...

stevexyz, 2020-07-01:
It was a suggestion to avoid to have a post modding in case, and from what I've understood then just switching the ROM would be enough.

Native ZX80 compatibility (an interesting machine on itself) would be surely a plus of this nice board, and your discrete logic approach is very similar to ZX80 philosophy!

mahjongg2, 2020-07-01:
I have not investigated what exactly the four pulldown resistors do, and I also fail to see what effect they would have on the Z80’s behaviour before reset. If I understand the trick they  somehow disable the initial NMI the z80 might see so that the software gets a chance to disable NMI’s before being forced into the NMI vector.

Don’t understand how that would work, and I would need conformation it does before adding new hardware to either the ZX81+35 or ZX81+38.

Especially the ZX81+35 is a closed chapter for me, as it was developed on an ancient laptop I hope I would never have to start up again.

Just had to create a new set of production files for the ZX81+35, and it was a very painful process.

stevexyz, 2020-08-28:
My suggestion was here for the plus38, that I consider a real beauty!

If the described change was working (I've anyway no possibility to test it but seems someone did) I was thinking of a jumper or a switch to switch ROM and in case other required circuit parts if needed, to have both the two computers into one (and the ZX80 is by far the fastest in basic ;))

Either way (with or without ZX80 compatibility) when you'll finish I want to try to build one for myself! :)

stevexyz, 2020-09-14:
The readme update states: "The use of a 4K ZX80 ROM is also supported", it means plus38 is working as a ZX80 even without hw changes (meaning just changing ROM)?

mahjongg2, 2020-09-14:
No that comment was only made because a ZX80 rom has a different pinout than a ZX81 ROM, and I have added jumpers so that a ZX80 rom would technically work in a ZX81+38 (its jumping can be changed so that the smaller ZX80 ROM can get power and data/address/control lines on the correct pins with some simple patches.

mahjongg2, 2020-09-14:
P.S. I have read the post that stated that you could Disable NMI even before any code runs by adding four pulldown resistors, and I see no reason why that trick would not work on my clone also, but that said "the proof of the pudding is in the eating", so it has to be tried out.

dkknab, 2021-03-05:
Got your zx81/80 board going tonight. Used 12mhz crystal though it wouldn't start consistently. Also used ZX80 rom loaded on AT28C256 making the changes per the board.  Couldn't get ZX81 rom to work... not sure why though it may be due to the lower crystal. Also couldn't get it to load from WAV files from phone on first try using ZX80 ROM.

mahjongg2, 2021-03-05:
Nice!

Do you have the latest PCB, if not it needs patches for slow mode!

To be hones, I have still not built my own board, as I’m quite disabled due to lockdown, my asthma has become crippling, and I’m very tired always. I found all the parts, I have the PCB,. and a simple but suitable soldering iron, but Revspace is completely closed, so I can go there.
I’m hoping to recover and get my strength back.

I’m wondering why the ZX81 rom doesn’t work, but that it works as a ZX80 is good news, I wasn’t certain of that.

Did you put the same code in the 16K ROM twice?. it needs that to be like an 8K ROM!

There is quite some activity round the ZX81 as many people are celebrating its 40th anniversary. I’m replaying to some posts with a link to my GitHub and revspace posts.

a 13MHz xtal should be necessary, as 12MHz, is quite far off for correct video timing! Farnell should have one!, also loading and saving can be affected, especially if you load original games and such.

keep trying, did you visit the german forum (if you can read German, I can) you can find it here: So sieht meine ZX81-Klonplatine aus <https://forum.tlienhard.com/phpBB3/viewtopic.php?f=2&t=3266>
various persons have gotten my ZX-81+38 to work, after these people found a few hardware bugs, I have made the changes.

there is also an english language discussion group discussing the ZX81+38, here: I started on my new ZX-81 replacement board (kit) ZX81+38 <https://www.sinclairzxworld.com/viewtopic.php?p=35160#p35160>

Good luck!

dkknab, 2021-03-05:
Thanks for the response. Let me see if I can share what I've done in
response to your questions. I will start by saying I am a complete novice.
I started with Ben Eater's 6502 project and have taken off from there
making 6502 boards and Z80 boards.

First off I am using the REV 1.6 board...the last one I believe you posted.
I got some from PCBway and JLCPCB. I'm not sure at all what you mean by the
patches for slow mode as I'm guessing it does not apply to REV1.6

I do have some extra boards - this was the first time I ordered a PCB so I
ordered some from PCBWay and JLCPCB. I have 10 boards in all so if you know
someone who needs one I can share it with them if that's ok with you.

Sorry to hear you are not 100%. I know for me working on these projects is
a distraction. I hope you get well soon!

I was thinking the reason the ZX81 rom doesn’t work is because of the
crystal. Once I get a 13mhz crystal I'll try the ZX81 rom again. It's neat
with the ZX80. I made the keyboard pictured on github. I didn't realize the
ZX80 keyboard is slightly different than the ZX81. I have two of the TIMEX
Sinclair 1000s that I've been messing with too.

I am using the AT28C256 eeprom which is 32K I believe. I only put in the
ZX81 code once. I'll have to load it a few times. Where do I start the
subsequent images of the code on the eeprom? Do I put it at 0K, then 8K,
16K and finally 24K locations?

I was completely surprised too that the 12MHz crystal works for the ZX80.
At least I know the board works just fine. I have ordered the 13mhz
crystal's from Mouser here in the US -I haven't tried to order from
Farnell. Where are they located?

AS I mentioned too I can't load from a audio file so the fact I have the
12mhz crsytal must be impacting that too.

I will keep trying. I do not read or speak German but maybe through the
wonder of the internet it will translate it for me. I'll be sure to check
it out as well as the english version of the discussion group you mentioned.

I'll let you know how things go especially when I get the new crystal and
if I have any luck with the ZX81 rom. Thanks for doing this!
David Knab

mahjongg2, 2021-03-05:
Dear David,

First I’m located in the Netherlands
and yes, the revision 1.6 Board is the latest, and should work fine, according to both the English and german forums, you can translate the german texts using copy/paste, and the google translate site.

I’m a bit shocked that 12MHz generates a good picture, I know that the original 6.5 Mhz resonators are not very precise, but 6 and 6.5 MHz, that is eight percent or so difference, and TV’s are very finicky about timing, even for a black and white signal.

I have been gifted a 1.6 PCB, so I have one, and a few older (1.4) versions too) I could patch up.

did you check the jumping for the AT28C256 (OTP) rom, as there are several jumpers that can be changed for older ROM’s, but AFAIK the default should be good for 32K ROM’s, and yes, I always put multiple copies, at 8K borders in, just so that there is less chance you put the 8K image at the wrong location in ROM, and yes at 8K borders, so 0K 8K 16K and 24K

I’m thinking about designing a specific keyboard PCB for the ZX81+38 too, although the ZX81+34 compatible keyboard works works fine.

Farnell has a website and operates worldwide.

best regards, Martin de Jong (mahjongg)

dkknab, 2021-03-05:
For the Atmel  AT28C256 rom, I did have to cut the trace on the underside
for pin 1. And I placed the jumper JP10 I believe correctly.  It may just
be me but I was struggling with how to set the jumpers because I couldn't
tell which pin is pin 1. Is it always the left pin when you read the text
properly?

The other jumper I set was JP11 as per your schematics too.

JP4 is also set for 60hz. I too was surprised 12mhz worked.  So does
12.288mhz but not on the first boot. I have to restart it a few times for
it to catch.

I'm not using only older roms. I like the fact that your board is made up
of readily available items.  That's what makes this so much easier and fun.
I'm trying to build the VIC2020... you may have seen this online.  The
problem there is you still need the VIC chip.  The fact that you got rid of
the ULA was a real enticement for me to do your board.

Take care

David Knab

dkknab, 2021-03-05:
PS the pcb board for the keyboard would probably take you no time to
design.  I wire wrapped mine on the blank pcb you see in the pictures.  It
works great on a sinclair time 1000 and with you zx81plus38 board too.

dkknab, 2021-03-05:
One last thing... next up I want to 3d print a case.  I'd love to do the
zx80 case in white since I have 2 timex 1000 cases in black.  I have the
creality ender v2 printer... talk about fun.  I've made all sorts of things
both useful and not with it.

David

dkknab, 2021-03-18:
Just thought I'd let you know I was able to get the zx81 rom loaded and
running.  The sound input to load programs also works. Here's a picture of
frogger... game over though.

David

mahjongg2, 2022-08-13:
well I finally got my own copy working, I had a revision 1.6 PCB too, and the changes that were needed were:
replacing the 74HC14 inverter with a 74HC04 inverter, as a crystal oscillator won't work with schmitt trigger inputs.
Replacing all logic IC's with 74HCxx logic, as a mix of 74LS and 74HC will often not work.
If your inverse K prompt is is all black then you also need the timing improvement provided by inverting the CPU clock using half of U18 the 74HC74, see revision 1.8 schematics. Also note that the order of pin's for the 5-pin keyboard data is reverted, fixed in revision 1.8. That might be relevant if you want to connect a real ZX-81 keyboard foil. as its almost impossible to twist the keyboard foil, so use a revision 1.8 PCB, BOM schematic, etc.

mahjongg2, 2022-08-23:
Just finished revision 1.9, the only changes from rev 1.8 is that R23 was changed to 1K and I added five pulldown 10K resistors on the bases of the 2N3904 transistors in my joystick interface, which should also work in ZX80 mode.

mahjongg2, 2023-06-13:
I just finished revision 1.10, the only changes are the location of some 2.5mm holes used to mount the PCB in the real zx-81 case.