mahjongg2/ZX81plus38 #3: RP1

ulysses66, 2020-09-27:
RP! on the schematics Rev1.4BW and BOM don't match.
Schematics show 4k7 and BOM shows 10k

mahjongg2, 2020-09-28:
Okay, it really doesn’t make much difference, as they are just pull ups, but in this case they correct value is 10K, because in the ZX81+35 used 10K resistors.

I will correct the schematic, and will re-upload the schematic.

Thanks for the tip

mahjongg2, 2020-10-20:
This is still an important change, with 4K7 pullups there were reports that pressing two keys simultaneously the ZX-81+38 could crash. So use 10K pullups!

dkknab, 2021-03-05:
Got your zx81/80 board going tonight.  Used 12mhz crystal though it wouldn't start consistently.  Also used ZX80 rom. Couldn't get ZX81 room to work.  Also couldn't get out to load from WAV files from phone on first try. Here's some pictures to show my progress.
[image]

dkknab, 2021-03-05:
[image]

mahjongg2, 2023-06-13:
if the 13MHz crystal doesn't work properly you might have used a 74HC14, that is incorrect as a 74HC14 uses schmitt trigger inputs that cannot be driven (well) with a crystal, so please replace the IC with a 74HC04, or better yet with a 74HCU04, and use the correct crystal the 13.000MHz one, or all (Video) timings will deviate.

Have fun with your ZX81+38.