mahjongg2/ZX81plus38 #5: Minor BOM issues

hackup, 2021-11-16:
First, thanks a lot for sharing this project! I'm currently in the process of building one for myself and it is a lot of fun.

I noticed two minor issues in the current BOM and I thought I'd let you know:
1. R20 is missing, according to the current schematics it should be 680ohms.
2. The Farnell code for RP1 is off, it was not updated when the value for RP1 changed.

Maybe I'll create a pull request in case that will help you fix this.

brochiman, 2021-12-13:
Hi and thanks. I think R21 is missing in the BOM, according to the current schematics it should be 100 ohms. Regards

hackup, 2021-12-13:
Yes, R21 is missing in the BOM, too. But it is just the series resistor for the optional external LED on the keyboard connector, right? So, I would say it is optional and its value is dependent on the type of LED you are using and on how bright you want it to shine.

brochiman, 2021-12-13:
oh well yes the value will depends about the LED. Thx regards

mahjongg2, 2022-01-07:
I agree, but for a 10mA red LED the value should be about 5V - 2V = 3V R=U/I 3V/0.01 = 300 Ohm, nearest resistor value is 330 Ohm.

closed 2022-01-07 by mahjongg2