mahjongg2/ZX81plus38 #10: Incomplete hires compatibility

squelch41, 2023-01-22:
Some software utilising hires modes works but not all
ROM mapped as 8k

WRX mode in particular seems prone to fail - examples
http://www.pictureviewerpro.com/hosting/zx81/fredn.htm
eg.
ZXTERM80

mahjongg2, 2023-01-23:
> Fred Nachbaur's ZX81Page

Dear Stefano.

It is disappointing to hear there is some software that doesn’t seem to work on the ZX81+38.
This is the first time I have heard about something not working on my clone.

I’m already investigating what is happening here.
I’m currently reading Fred Nachbaur’s ZX81Page to find out, and have downloaded the ZX*TERM*80 file which contained a text and some .P files, one of which was the sam text as the readable file, and the other .P file contained probably the emulator file. However the text made me no wiser.

I’m also investigating what exactly WRX is and whether MOD is some hardware extension (MOD for MODEM?) as it seems that for any terminal program you will need a serial interface, and a modem.

Ill try to go to the bottom of this, and when there is something I can change in the design of the ZX81+38 I will.

I will also put a warning on my GitHub page to inform builders of my clone that this particular highness package (as I assume that it attempts to create 80-column text on a ZX-81, which needs some very smart video generation software?!) won’t work on my clone.

P.S. What do you mean by Issue #10?

also what exactly happens when you try to run this software package? Could you try another program that uses WRX, like 3-D fractals? and let me know if that works.
The description given of ZX*TERM*80 that it needs external hardware " For one thing, it's hardware dependent (either Byte-Back or Timex/Westridge modem). For another, the hardware is long gone.” so did you add this hardware?

I do own a real ZX-81 but it do not have a 16K expansion, so I cannot try to run this software myself on the real ZX-81, but as I’m going to design a hardware extension that has 16/32K of RAM, an 8K ROM expansion, a ZON compatible PSG with an AY-3-8912 and a SD-card reader (with firmware in ROM) I might be able to try ZXTERM80 at a later opportunity.

Thanks for letting me know this, if I find any news about this I will let you know. I’m known for my stubborness so be sure I won’t stop until I know act is behind this issue!

With best regards

Martin de Jong.

mahjongg2, 2023-01-23:
Stefano,

It also mentions it only runs with another clone creators “wilf board”, I do know will richter (from his designs), and will try to find out what He did to get ZXTERM80 to run.

best regard, Martin de jong.

mahjongg2, 2023-01-23:
A few remarks.

The description on the treasure chest page of fred nachbaur's page on pictureviewerpro.com said that ZX*TERM*80 only works with a "wilf board", So I tried to contact wilf rigger, but the e-mail address rigter@cafe.net is no longer in use.
I found an article about the WRX16 code here: https://archive.org/details/zx-appeal/ZX-Appeal%20Jan%2086/page/12/mode/2up?view=theater     and the article states that this code uses RAM at 8 to 16K which my clone supports, but note that it is a copy of RAM at another address, so may not work. Also the article states that you need a Z80 that uses 8-bit refresh's counter, I do now (from experience) that normally Z80-s have a 7-bit refresh counter, that may be one additional problem. I'm not sure modern (C-MOS) Z80's have adopted 8-bit refresh counters, but they might not!
If someone can give me a valid e-mail address for wilf rigter I will try again to contact him.

Martin de Jong aka mahjongg.

mahjongg2, 2023-01-23:
I will try the four .zip files on my own ZX81+38 clone.
I do have a working (real) ZX-81, but not a 16K expansion board.

mahjongg2, 2023-01-23:
sorry closed by accident now re-opened

squelch41, 2023-01-23:
I do have a serial interface board that acts like a westridge modem designed by wilf rigter
https://web.archive.org/web/20100121145915/http://www.user.dccnet.com/wrigter/index_files/serial.htm.
This works with the zx81+38 using the westridge modem software 

[Westridge terminal Software MTERM.zip](https://github.com/mahjongg2/ZX81plus38/files/10483789/Westridge.terminal.Software.MTERM.zip)

Screenshot from loading zx80term when it initialises hires mode:
[image]

vs in EightOne emulator with ram at 8k-16k option enabled
[image]

I will try 3d-fractals
The other wrx high res programs attached in the inital issue report do not work on the zx81+38 but do in the EightyOne emulator.

[ZX-TERM-80.pdf](https://github.com/mahjongg2/ZX81plus38/files/10483894/ZX-TERM-80.pdf)

squelch41, 2023-01-23:
Looks like the z80 processor in the BOM does support 8-bit refresh, looking at the data sheet
Z84C0008PEG
https://eu.mouser.com/datasheet/2/240/ps0178-2584834.pdf

mahjongg2, 2023-01-24:
it does use 8-bit refresh?! Well, okay, that is one less worry, unless a builder chooses not to use a new Z84C0008PEG but uses an old Zx80, but let's now worry about that just yet.
I did again mail wilf rigter with a new e-mail address (wrichter@dccnet.com) and unlike my previous attempt this mail has not bounced, but I have got no reply yet. I also contacted a compatriot (I'm Dutch) named Rodney Knaap, who also built ZX-81 zx-81 clones in the past, and I assumed knew wilf. Well he did band he replied but he stopped doing anything related to the ZX-81, and it doesn't seem likely he can offer any assistance, but he did give me the @dccnet e-mail address. so there is that.
My educated guess is still that the 8K RAM at $2000 to $2FFF is a problem, as it is **not** indipendant from RAM used by the ZX-81 itself. I still have not made a diagram where this RAM is mirrored, but my intuition (working this out in my picture thinkers brain) tells me it is mirrored on the second 8K block of the ZX-81 16K RAM the ZX-81 uses, that is RAM at $8000 to $BFFF range so if the WRX16 code writes its high res bit pattern into RAM at $2000 it will overwrite code in the range of $8000 to $BFFF, and this will cause the code to crash. For the WRX16 code to work it needs (as wilf clearly said when he published the code) it needs an **extra** 8K of ram at $2000 to $2FFF and the ZX81+38 does not offer an independent block of 8K of RAM (not even when you change the ROM address block to 8K instead of the standard 16K it is now configured as. 

I don't know immediately how I could change the address decoding to do that. Now the decoding is simply to give ROM 8K afrom $0000 to $1FFF and enable the SRAM chip for all other addresses! Then the actual RAM chip can be wired to occupy 16K or 32K of RAM.

I will try to explain the conundrum to wilf rigter, if he replies, and still want to help, perhaps he has an idea why his high-resolution system won't work on my system.

to be continued.

squelch41, 2023-01-24:
Tested a few more of those programs
3D fractal doesnt work on EightyOne emulator or zx81+38 - loads but run try and run just crashes at basic at line 4

Macro-life 16k runs in EightOne emulator but on zx81+38
[image]

mahjongg2, 2023-01-24:
my current explanation why WRX16 doesn’t work on a ZX81+38 is that it actually doesn’t have the required 8K RAM at $2000, not even when you switch to 8K ROM space, instead of the default 16K ROM space.

True when using 8K ROM space there is RAM at $2000 but it is NOT independent, it is mirrored into the working RAM space, so if WRX16 writes in the RAM at $2000 it is also affecting the RAM at the area used as working RAM, and overwrites the program code there.

To get WRX16 based code to work, you have to add a separate 8K of RAM that disables the 16/32K RAM chip when accessed, and is located at $2000.

I have no idea how I can change the ram chip select decoder so that a separate piece of 8K RAM is placed at $2000 so I now are of the conviction that WRX16 cannot work on my ZX81+38, and in the past it couldn’t work on an unchanged ZX-81 (plus 16K RAM pack) either, you needed to add an independent 8K of RAM (and maybe replaced the Z80 by a more modern one with 8-bit refresh).

The documentation of the ZX-TERM*80 (ZX-TERM-80.PDF) has this to say about it. “notes, even with 64K RAM, you will still need a CMOS (static) RAM in the 8-16K region”. And yes eighty one will work, but that is only because it has a setting that emulates that independent piece of 8K RAM

so sorry, but the only solution I see, is to create an 8K RAM adapter (expansion board) you can plug into your ZX-81 or ZX81+38.

Sorry, but if you have a better idea please don’t hesitate to contact me. I’m also writing in the gitHub ZX81+38 page about this...

with best regards, Martin de jong,

p.s. I’m still waiting for Wilf Rigter to reply, I have contacted another ZX-81 expert Rodney Knaap, but he could not help me either than to give me a real e-mail address for wilf, it hasn’t bounced yet, as the previous did within minutes.

squelch41, 2023-01-24:
Following the modification thread, there is a bit of information here about ram pack modification for WRX but, to be honest, I dont understand it!
https://sinclairzxworld.com/viewtopic.php?t=1737

mahjongg2, 2023-01-24:
Thanks Stephan, I will have a look at it, and see if it changes anything….

Hopefully I will “understand it”, LOL.
But I guess its just a modification to RAM pack logic, it might not relate to the RAM in the zx81+38….

the best. Martin.

mahjongg2, 2023-01-25:
As I understand it the link goes to a forum I often visited and  have use used to talk about my clones, I have gotten quite some bug fixes from there. And from a german ZX-81 forum.
But what they are discussing here is a new way of fixing a 16K RAM cartridge so it will work with WRZ16, it seems you needed to patch your RAM pack at the time by adding two diodes and a resistor (adding a diode NOR) . But the forum post is about someone who thought he could patch the RAM cartridge _without_ using any components. The (often technically well informed) users of the forum told him that his solution would be unreliable. It seems that a RAM pack had an internal refresh counter, and the diode NOR patch was needed to make that working correctly, or ram content could get lost by lack of refreshing it.
I don't think it has any relevance to my ZX81+38, that I can see.

Also I have had no reply from Wilf Rigter yet.....

So currently my unfortunate conclusion is that to use WRX16 on my ZX-81 clone is the same as for a real ZX-81 is that you will need extensive modification for it to work.

mahjongg2, 2023-01-26:
I learned there are several divergent "variances" of the WRX16 video driver, I suggest creating another one, specifically for the ZX81+38 (and other ZX-81's with 32K of RAM).

Instead of using low RAM (RAM located directly after the ROM, so from $2000 to $2fff ) I suggest rewriting the code so that it uses RAM directly after the 16K RAM instead of 8K low RAM.

Then jumpering the SRAM chip so that it occupies 32K, and there will be RAM from $8000 to $FFFF.

I suggest naming this variant WRX32, as it uses the extension of RAM to 32K.

This version can then be combined with software like ZX-TERMINAL*80 and other software that now uses the WRX16 driver to create high resolution graphics.

squelch41, 2023-01-26:
Thanks
This article seems to contain some interesting explanation of how wrx works
Will see if can make sense of it

https://archive.org/details/SyncWareNews/SyncWare%20News%20v4n2/page/16/mode/1up?view=theater

mahjongg2, 2023-01-27:
I have glanced through it, but for now it only strengthens my opinion that WRX16 NEEDS EXTRA static bank of RAM at 8-16K and I cannot simply provide that.
There is talk of a two diode plus pullup gate that does something special for the RAM. but I don’t see the need for it if we don’t have extra RAM.

True the complete functioning of WRX16 goes over my head, I don’t understand the whole thing yet.
Perhaps someone can make a version of WRX16 that doesn’t need independent SRAM at 8K-16K, but uses the extra RAM that the 32K RAM chip can provide, if that is the case, and the only hardware patch needed is the diode gate, I’m willing to add it to revision 2.0,

let me know what you can find out.

Thanks for this info….

martin.

mahjongg2, 2023-01-27:
I have downloaded Syncware News v4n2.pd so that I can read it at my leisure, again thanks for finding the info.

squelch41, 2023-01-28:
Again, dont know if this is relevant to the design of the zx81+38 but there was a 32k sram that replaced the internal ram of the real zx81 and had wiring that enabled it to run wrx

https://forum.system-cfg.com/viewtopic.php?f=18&t=11512

Seems to involve some changes at a14 and a15 gated by M1

squelch41, 2026-11-14:
I think this is now fixed-

This board modification fixed it along with setting JP12 to 32k

[image]

I did use claude to help troubleshoot.

Report below.

## WRX16 true hi-res working on ZX81+38 rev 1.9 (two changes needed)

WRX16 programs that put the bitmap in 8-16K (I register 0x20-0x3F) now run on my board. It needed two changes, not one.

### 1. RAM mapping: JP12 to A14 (32K)
With JP12 at the default 16K, the SRAM ignores A14 and A15, so 0x2000-0x3FFF is the same RAM as 0x6000-0x7FFF, and a bitmap there overwrites the program and system area. With JP12 on A14 there is a separate bank at 0x2000-0x3FFF (aliased at 0xA000-0xBFFF) and at 0x8000-0x9FFF. JP1 can stay at the default 16K; `POKE 8192,55: PRINT PEEK 8192` returns 55. A memory test (CHECKMEM.P) reports 0-8K ROM, 8-40K RAM, and mirrors above 40K.

### 2. Lookup-disable gate: include A13
U12D computes `A14 OR nRFSH` and drives the /OE of U4 (74HC574) and the enable of U6A (74HC125), which put character code and row onto A0'-A8' during refresh. True hires needs that lookup disabled so the CPU's I:R address reaches the RAM. As built, only I >= 0x40 (A14 high) does that. I = 0x20-0x3F (A13 high, A14 low), where WRX16 puts its 8K RAM, still gets the pattern lookup and shows a repeating pattern instead of the bitmap. Wilf Rigter's ZX97 notes disable the lookup when A13 or A14 is high.

**Fix:** cut the A14 trace to U12 pin 12 and drive pin 12 from A13 OR A14. I used two 1N4148 diodes (anodes to Z80 pin 3 and pin 4, cathodes joined at pin 12) and a 4.7K pull-down to GND. For a future revision, the spare U12C OR gate (pins 9 and 10 are currently grounded, output pin 8) could do this with no extra parts.

### Evidence
| Program | I range | Result |
|---|---|---|
| Rigter WRX-16 Mini Demo | 0x41-0x42 | works before and after |
| Defender (DFNDA.P) | 0x74-0x7D, patterns 0x80-0x9E | failed with JP12 at 16K, works with JP12 on A14 |
| Mini Demo patched to bitmap at 0x6100 / 0x2100 | 0x61 / 0x21 | before gate fix: 0x6100 shows bars, 0x2100 shows the lookup-mode result; an emulator shows bars for both. After: Both show the bars |
| 194TWO, ZX-Term*80 v2.5 | 0x20-0x35 | failed before, works after|

WRX2000 and WRX6000 test programs - modified versions of Wilf Rigters WRX-16 Mini demo.
[WRXTEST.zip](https://github.com/user-attachments/files/34851718/WRXTEST.zip)