mahjongg2/ZX81plus38 #11: Revision 2.0

ghhann, 2023-06-06:
Thanks for doing this project.  I read that you now have a revision 2.0.   Anyway to get the files to send to pcbway or are you listing it as a shared project

All my best

mahjongg2, 2023-06-13:
Its not revision 2.0 but revision 1.10, there should be no electrical differences with revision 1.9, just the positions of a few 2.5mm drill holes for mounting the PCB to a real ZX-81 plastic enclosure were corrected, so the ZX81+38 PCB will now definitely fit into the real enclosure. So even though the schematic is now also called rev 1.10, it is exactly the same as revision 1.9.
I also added the KiCAD files for revision 1.10, so if anyone ever wants to continue with my design, please use revision 1.10.
I'm in contact with PCBWay, and will try to list my revision 1.10 design as a shared project.
will mention that here if it happens.

ghhann, 2023-06-13:
Thank you I was planning on using a real case.  Thanks!

closed 2023-06-13 by ghhann

mahjongg2, 2023-06-13:
added one comment, BOM revision 1.10