mahjongg2/ZX81plus38 #13: Using DIP-24 ROM

TankedThomas, 2023-07-25:
Hi Martin,

Awesome project, thanks so much.

Just a quick question: is it possible to use an original 24-pin ROM chip? 
A local seller has a bunch of new old stock parts so I thought it could be a bit of fun instead of flashing my own. 
Obviously it's not a drop-in replacement for a modern 28-pin EPROM though, so it would need an adapter.

Is it the same pinout as a DIP-24 to DIP-28 ROM adapter for the C64 or is it something that would need to be made from scratch?
Something like this is what I'm referring to: [https://github.com/SukkoPera/Open2327RomAdapter](https://github.com/SukkoPera/Open2327RomAdapter)

mahjongg2, 2023-08-29:
Dear Stephano,

a bit late, and was a bit confused about your question.

My ZX81+39 board supports the use of the original 24 pin ROM, just change the applicable jumpers for a 24 PIN ROM, look in the schematics.
and place the 24-pin ROM in the lower part of the socket, so do not use pin 1 , 2, 28 and 27

If you source new old stock ROM’s make sure you buy a ROM that contains the code for a ZX-81, not any other ROM, there must be thousands of old ROM’s that contain something else than the ZX-81 code.

the adapter you are talking about is so you can replace a 24-pin ROM  with a 27Cxxx style EPROM, and this isn’t relevant in this case

best regards, martin de jong.

TankedThomas, 2023-09-04:
Thanks Martin, I'll have a look at the jumpers. I assumed I'd need an adapter instead as the board defaults to 27C-style ROMs.

The documentation for them is a bit vague and I just had a burnout from trying to use the 32K RAM jumper, so I'm not in a hurry to use the rest.

I did, however, install jumpers to use a 27C64. Based on my knowledge of the ROMs, it should be enabled by using jumpers JP9 and JP10 to pull the necessary address lines high.

FYI, the new old stock ROM I got is a genuine Sinclair ZX-81 original.

closed 2023-09-04 by TankedThomas