mahjongg2/ZX81plus38 #14: Not Compatible with ZX81 USA Cases

TankedThomas, 2023-08-06:
I probably should have checked this beforehand but the project description mentions the ZX81+38 is designed to work with an original ZX81 case.

However, now that I have the PCB in-hand, I noticed that it doesn't fit an original ZX81 USA model case.
The biggest issues are that the composite jack is in the wrong place, though the lower left screw hole is misaligned too (perhaps some of the others, but that's the main one).

I did a bit of reading and it turns out that they used the same case but in-filled one slot depending on if it was a UK or USA release, because of the way the RF modulator faces. The US modulator was VHF, with the jack towards the rear, and the UK modulator was UHF, with the jack right next to the other three.
Interestingly, my case is from New Zealand but we only had VHF in the 80s as far as I recall, so it seems we got the USA case and modulator with the UK keyboard, and it still says "ZX81 USA" on the underside. I would have to research this further but it seems likely, then, that Australia and New Zealand both got the USA case.

I realise it's not particularly easy, especially given the additional logic chips to make up for the lack of the ULA, but I was wondering if it would be possible to redesign the PCB slightly to allow the composite jack in either position, so that it can fit both cases, especially since they seem to be more common than I would have expected. This would probably allow the PCB to then fit a Timex 1000 case too.

mahjongg2, 2023-08-26:
Dear  Thomas,

unfortunately I have to disappoint you. I have looked at your proposal to make a version of the ZX81+38 that is compatible with the enclosure of both the original ZX-81, and the Timex Sinclair 1000, and I do not see a solution with two “RF” plugs, there simply isn’t place for that!. I also have investigated moving CO5 out of the way, and using the freed space to moving U21 and the crystal and other components of the crystal oscillator there (thus making a specific TS 1000 version), but that might be possible, but I don’t think it is a good idea. I suggest just modifying the case of your TS-1000.

As for the lower left hole screw, it and also for a few other screw holes, these are corrected in the revision 1.10 version, in fact the only change between revision 1.9 and 1.10 are the locations of these holes.

so in short, not it is NOT possible to create a PCB that can fit a TS-1000 case too, and no I won’t create a specific TS-1000 version. Compared to the original only very few Timex models were made.

I’m busy designing a mechanical keyboard, much larger than my original that is the same size as the original, so I expect that this will lead to new enclosures, either made out of wood, or somehow 3D printed. That might be an alternative to using a TS-1000 case. Also note that if you want to use the joystick port you cannot escape drilling holes in the case.

I’m sorry, but this is my decision.
With best regards, Martin de Jong.

TankedThomas, 2023-09-04:
Hi Martin,

That's fine, I understand.

It is worth noting, however, that this isn't just for the Timex TS-1000 but, as I mentioned, the New Zealand (and, I assume, Australian) ZX-81s, as well as the ZX-81 USA edition. How many of these were made though, I'm not sure.

As far as the screw holes go, I used the rev 1.10 gerbers you provided, so I'm not sure why they don't align.

mahjongg2, 2023-10-18:
Dear Thomas, well I thought it through, and decided that it is worth the trouble to create a specific version of the ZX81+38 for the TS1000, I will name this version the TS1000+41.
It will appear when finished as a separate project.

mahjongg2, 2023-11-13:
well, I did it, I created a new project called TS1000plus41, you should be able to find it here: https://github.com/mahjongg2/TS1000plus41
you can find the schematic (which is the same except that JP4 is bridged for NTSC video as default), all the files to start your own KiCad, gerber and drill files in a .ZIP for creating a PCB, even its own BOM, what is missing is a component overview drawing, but not much has changed, I will create a new drawing for the TS1000plus41. I did NOT attempt to move a drill hole, as I have no idea which drill hole you meant, and in which direction it should move.