mahjongg2/ZX81plus38 #15: A0' - A8' at expansion port

dikdom, 2023-10-25:
Dear Martin,

this is an excellent (and tremendous) piece of work you made.
What I would like to ask what was behind the decision that you changed the address lines on the expansion port compared to a stock ZX81? You routed there the A0' - A8' which are the address lines after a resistor instead of the A0-A8 CPU addr lines. 

I'm building an expansion cart for the ZX81 (it has 32kRAM, 64k EEPROM, 8M flash, SD card slot and a cursor joy port).
On a stock ZX81 the WRX mode works properly.
I'm thinking (unfortunately didn't get there yet to test) if would the WRX mode work with the +38 board, completely from the cart's RAM (yeah, my cart disables the internal RAM). 

Thanks!

PPZ-00, 2024-06-09:
Hello, Is there any workaround for that? I have built the ZX81 Plus 38, in the beginning I thought that I did something wrong :) 
Now I read it's the design flaw.

dikdom, 2024-06-10:
Hello,
it turned out that my design was not a good one...
So what I do is that I have a GAL chip on my cart and it
- disables internal RAM
- enables external RAM on $4000-$FFFF access, except $C000-$FFFF read on /M1 phase
- enables external ROM on $0000-$3FFF read access, except /RFSH phase. This is the case when the ULA reads the character info from the internal ROM.
So originally I enabled this internal ROM access only when /RFSH and /MREQ both were active (low). That didn't work with +38. When I removed the /MREQ from the GAL's condition table (so the internal ROM was enabled for the whole /RFSH cycle) then the +38 worked well. It seemed that it works pretty well with stock ZX81, so I made ~40 pcs of cart for fellow ZX81'ers. 
The fun started here: it turned out that there are problems with specific ULA+ROM chip types. The problem was gone, when the GAL condition was reset to the stricter one (both /RFSH and /MREQ is low). 
So now I have two GAL firmware codes. I need to doublecheck with the TS1000/TS1500 which of the firmware they are OK with.

dikdom, 2024-06-10:
> Hello, Is there any workaround for that? I have built the ZX81 Plus 38, in the beginning I thought that I did something wrong :) Now I read it's the design flaw.

Uh-oh. Now I realized that you're asking about the +38. Well.. I built a +38 and it properly works for me. (without my stupid cart)

PPZ-00, 2024-06-10:
Your cart is great, as well as Plus 38 :) 
Great designs, both of them. 
I built two of Plus 38, and 5 of your carts (I will keep two of them) but when using it with Plus 38, I cannot start some software, I get garbled screen. 
So I thought that I did something (I initially installed nmos Z80 CPU in one of them, but quickly realized that it will not work, even though it boots to a prompt).
Now I can't run some of 32k programs from your demo dump for eeprom, and that is actually the only problem. 
So I thought naturally that I did something wrong in assembling the Plus 38. 
Anyway, even if this can not be fixed, it's a great project, and your cart is awesome. 
Although a bit expensive to build because of the two GALs (I used ATF22V10 and ATF20V8, ot works great).