mahjongg2/ZX81plus38 #16: Keyboard production files

dikdom, 2023-11-28:
1 - in the file name it is stated it's v1.2, on the top copper it is written rev 1.1
2 - there is no bottom copper and silk layer, so TH components cannot be soldered properly..

mahjongg2, 2023-11-29:
okay, I will check your points, and will repair them, give me a few days.

dikdom, 2023-12-01:
> okay, I will check your points, and will repair them, give me a few days.

Thanks. 
In the meantime I've read the README and that pcb is shown rev1.1 because no tracing layout changed between 1.1 and 1.2, just the silk. Not a fortunate thing, because visibly there is change
The bottom layer is missing, I checked it with the JLC's gerber view and produced that way...

mahjongg2, 2023-12-01:
Okay, I updated the .ZIP with the keypad files, the schematic now has references to revision 1,2 and today as date, and explains that the schematic stays the same as revision 1.1.
the layout (kicad PCB) kicad file now has revision 1.2 written in copper (top side)
note the IS NO BOTTOM COPPER SIDE!, this is a single sided PCB, that happens to have its copper on the TOP SIDE ONLY, and has silk screen text on the copper (top) side also. 
I hope this is now clear!
Note that if the PCB manufacturer mistakenly puts the copper on the bottom side, and the silk screen on the top side then the PCB is still useable, and you can solder the pins of the button on the bottom side instead of on the top side, but mechanically this isn't as sound as soldering the pins on the top side, prolonged pressing on the keys may cause the traces to come unglued. This will not happen as easily when he soldering is done on the top side. so that the copper traces are (only) on the top side is by design! Also because you can stick the connection wires in through the bottom, and solder them on top is the better option!

I hope this clarifies the design choices made for this keyboard PCB.

mahjongg2, 2023-12-01:
Dear stephano, i have update the .zip with KiCAD files,  and the schematic.

There is no bottom layer! Because it is a single sided PCB, with the copper layer (and the silk screen etc) on the TOP LAYER ONLY!

ihis is per design as this way pressing on the button presses the traces towards the PCB, that is much stronger than pressing the buttons off the PCB, and the wiring can be soldered on top of the PCB, so the wires protrude through the bottom.

So no the bottom layer is NOT missing!

best regards, martin de jong.